spiritualized1997/openFPGA-GBA #1: Source Files?

atrac17, 2022-07-29:
Where is the source?

Goldanas, 2022-07-29:
It's in the releases

atrac17, 2022-07-29:
> It's in the releases

The source verilog is in releases? What files?

Goldanas, 2022-07-29:
Ah, you're right, totally not there, my apologies

birdybro, 2022-07-29:
I would like to see the source files for this Open FPGA distribution. I'm assuming "Open" implies it's open source, does it not?

atrac17, 2022-07-29:
> I would like to see the source files for this Open FPGA distribution. I'm assuming "Open" implies it's open source, does it not?

Good luck with that happening, bots and new accounts... I mean, everyone knows this is kevtris' burner for a jailbreak. Rumble support after an hour of the SDK coming out? Yeah, okay.

reminon, 2022-07-30:
They also mentioned that it's been in testing for months.

rlaphoenix, 2022-07-31:
> It's in the releases

Even if the data is there, it should be in the repository so we can compare changes easier between releases, to actually benefit from `git` to begin with.

coolbho3k, 2022-07-31:
It's a core _for_ their framework called openFPGA, which does not necessarily imply that the core itself is in any way open. This is almost certainly a descendent of the Pocket's built-in GBA core. Analogue is not going to release that.

dogtopus, 2022-08-11:
I see fake open source that wastes GitHub server resource I move away. Still won't buy any Analogue stuff until there's some real commitment.