spiritualized1997/openFPGA-GBA #2: WarioWare Twisted

elgarlic, 2022-07-29:
Since this is mentioned in the description, I thought I would try out the US rom while the Japanese (affordable) cartridge is inserted. No gyro function is working. Should it be? Thanks!

RewoundVHS, 2022-07-29:
It is explicitly stated that WarioWare Twisted will not work because the Analogue Pocket lacks a gyroscope.

elgarlic, 2022-07-29:
What it explicitly says is "Warioware Twisted would be if there was a usable accelerometer...". Obviously, it does not specify what is considered usable. I provided an accelerometer, and was wondering if that should be sufficient. If the core can't access the cart slot, that's perfectly understandable.

hiddenindians, 2022-07-29:
Would the DS Motion Sensor (The one shipped with Tony Hawk Motion) work as a proxy gyro?

HeyItsLollie, 2022-07-31:
Could gamepads like the DualShock 4 or DualSense be viable options for gyro/accelerometers, while playing via the Dock?

Mi5KL, 2023-02-06:
Yeah, I've been wondering this too. If one game's gyro stuff can extend to the whole set then it would mean you could pick up a super cheap copy of Korokoro Puzzle Happy Panechu and be able to play the english versions of Yoshi and WarioWare, though I'm not really sure if they all work exactly the same.