spiritualized1997/openFPGA-GBA #4: Possible refreshing issue in Pokémon Pinball

rlaphoenix, 2022-07-29:
https://user-images.githubusercontent.com/17136956/181850808-603ac44e-99ab-42da-8fa7-0fe5bbeaa751.mp4

It's much more noticeable in person. Notice some pixels get darker (or brighter) frame by frame.

[image][image][image]

dotdissent, 2022-07-30:
Are you referring to the compressed mess of pixels at the bottom of those images? That's an artifact exclusive to the European version of the game. Try the US version.

rlaphoenix, 2022-07-30:
> Are you referring to the compressed mess of pixels at the bottom of those images? That's an artifact exclusive to the European version of the game. Try the US version.

That's not specifically what I'm referring to, no. But it does not seem to happen or at least as much on the US version.

The flickering or judder of the fade to white is still happening. Maybe its just my pocket? I'm not sure.

I've captured it in slow motion 240 fps, and you can see it redraws in a weird way, its not consistent across the screen, nor is the timing. This is probably why I perceive a judder on playback.

Slow-motion:

https://user-images.githubusercontent.com/17136956/181880261-408345a0-acef-4881-8bf0-b645009ebd99.mp4

Real-time:

https://user-images.githubusercontent.com/17136956/181880285-bb4ba75c-6344-4e74-b51b-b646d288507c.mp4

SegaSnatcher, 2022-07-30:
Here's what it looks like on a OG GBA with an IPS screen.

https://www.youtube.com/watch?v=6ZhrQxAvhmw

rlaphoenix, 2022-07-31:
After checking around it seems to be somewhat intended/normal for this issue to happen on IPS or otherwise non-original display hardware. As for why it refreshes each frame weirdly in my slow motion video, that may be related to just the camera and screen refresh rate not matching correctly.

This is likely not an issue with the core, apologies.

closed 2022-07-31 by rlaphoenix