spiritualized1997/openFPGA-GBA #11: Pokemon GS Chronicles (fan Rom) won't load

PriMieon, 2022-07-30:
I have tried the following fan games:
Pokemon unbound (GBA)
Pokemon prism (GBC)
Pokemon GS Chronicles (GBA) 

All of them with except GS Chronicles work. GS Chronicles gets stuck on GBA loading screen
The animation plays and then it stays there

rlaphoenix, 2022-07-31:
I wonder if it could be a bad ROM, maybe a bad IPS patch, or a bad BIOS?

GreatHierophant, 2022-07-31:
I found the patch for Pokemon GS Chronicles and applied it to the correct ROM, Pokemon - FireRed Version (USA).gba.  I found that the hack did not load in my EverDrive GBA on a GBA SP and did not load in mGBA until I chose not to load the BIOS.  They should update their patch to work with original hardware.

rlaphoenix, 2022-07-31:
Have you verified the ROM dump of FireRed as good? Does it match No-Intro? This is strange that it could be to do with the BIOS based on your test on real hardware and mGBA.

GreatHierophant, 2022-07-31:
The patch is a .bps patch and the patching program reported the ROM patched successfully with the no-rev FireRed USA ROM and failed with the rev 1 FireRed USA ROM and I have verified that the crc32s of these ROMs match the no-intro entries.

It is not so much strange as just that this hack was never tested with original hardware.  Such hacks are unfortunately too common.

PriMieon, 2022-07-31:
I have tested the rom on my ez flash 
The color of the game overall seems to be very dark and when pressing the start menu the game freezes for about half a minute (but music will play. Just it won't accept inputs for 30 seconds)

So the rom works with a few imperfections on my ez flash. Won't work on the analogue though.