spiritualized1997/openFPGA-GBA #12: This is not an issue, I want to help.

Dnilo, 2022-07-30:
I'm a pixel artist and would like to contribute by improving the pixel icons on the core menus, I'm just not familiar with GitHub. 

My apologies in advance for this entry.

KeenIIDX, 2022-08-01:
The ‘Jobs’ link at the bottom of Analogue’s Store page just links to the email [info@analogue.co](mailto:info@analogue.co?subject=Jobs%20at%20Analogue), so shoot them your resume.

Dnilo, 2022-08-01:
> The ‘Jobs’ link at the bottom of Analogue’s Store page just links to the email [info@analogue.co](mailto:info@analogue.co?subject=Jobs%20at%20Analogue), so shoot them your resume.

I'm sorry but... what does Analogue have to do with anything here?

JackWitherell, 2022-08-01:
You should be able to on here
- fork this repo

and then use a git client on your computer to
- log into your github account
- pull the forked repo that you made to your computer (the download button on github gives a url you can use to pull, or your git client will list your repos assigned to your account)
- edit the files in the pulled repo on your machine
- commit the changes locally
- push the local repo's commits to your github account
- create a pull request on this repo with your change (the new PR button will show up once you've committed the changes into your forked repo)
- have your pull request reviewed and merged by the owner of the repo or a contributor with permissions.

It might sound very complicated from an artist's perspective but this is in fact the way that bespoke programmers have been using git for decades.

nateleavitt, 2022-08-26:
I am also curious to know if the source code is available, because I don't see any in this repo. Would love to contribute.

smantic, 2024-12-20:
> You should be able to, on github
> 
>     * fork this repo
> 
> 
> and then use a git client on your computer to
> 
>     * log into your github account
> 
>     * pull the forked repo that you made to your computer (the download button on github gives a url you can use to pull, or your git client will list your repos assigned to your account)
> 
>     * edit the files in the pulled repo on your machine
> 
>     * commit the changes locally using your git client
> 
>     * push the local repo's commits to your github account using your git client
> 
> 
> and then finally
> 
>     * create a pull request on this repo with your change (the new PR button will show up once you've committed the changes into your forked repo)
> 
>     * have your pull request reviewed and merged by the owner of the repo or a contributor with permissions.
> 
> 
> It might sound very complicated from an artist's perspective but this is in fact the way that bespoke programmers have been using git for decades.
> 
> I'd recommend closing this issue. More information on helping with the project can be found elsewhere and is not specific to this repo.

there is nothing in the repo to edit

JackWitherell, 2025-03-08:
> [@JackWitherell](https://github.com/JackWitherell) there is nothing in the repo to edit

Yeah that's been mentioned, I got it now thanks.