spiritualized1997/openFPGA-GBA #7: Volume is reduced compared to Official Pocket GBA core.

SegaSnatcher, 2022-07-30:
Here are video comparisons of the Official Pocket GBA core at max volume vs the OpenFPGA GBA core at max volume.

https://www.youtube.com/watch?v=Nb5l3qiFotQ

https://www.youtube.com/watch?v=2bl0f6a4mtA

PriMieon, 2022-08-01:
I find it not too apparent in the videos but it is super noticeable in person! Huge decrease in volume

rootjam, 2022-08-06:
i am having the same problem only GBA sound is low color and regular Gameboy are fine. I Put in my original cart and it blows the sound out of the water compared to the OpenFPGA

SegaSnatcher, 2022-08-07:
> i am having the same problem only GBA sound is low color and regular Gameboy are fine. I Put in my original cart and it blows the sound out of the water compared to the OpenFPGA

The GB/GBC OpenFPGA volumes are lower as well vs using real carts or .Pocket files.

PriMieon, 2022-08-07:
> > i am having the same problem only GBA sound is low color and regular Gameboy are fine. I Put in my original cart and it blows the sound out of the water compared to the OpenFPGA
> 
> The GB/GBC OpenFPGA volumes are lower as well vs using real carts or .Pocket files.

For sure. The volume is a lot quieter with the non-advance consoles

SegaSnatcher, 2022-08-23:
1.1-beta-2 that was rolled out today fixes the volume limit issues.

closed 2022-08-23 by SegaSnatcher