spiritualized1997/openFPGA-GBA #14: Display Modes don't work

ifighftdragons, 2022-07-31:
Display Modes don't work

SegaSnatcher, 2022-07-31:
This is known, developer is waiting for an API update.

tommy, 2024-01-14:
The [2.0 firmware update](https://www.analogue.co/support/pocket/firmware/2.0) added display mode support for openFPGA cores. Is there any plan to add support for the Spiritualized GBA core?

v-kiniv, 2024-02-20:
On fw >=2.0 you can enable `Original GBA LCD` and `Original GBA SP 101` display modes by modifying the `Cores/Spiritualized.GBA/video.json` file and adding corresponding modes to `display_modes` section. Update to the core itself is not required. Here's an example of the modified file:

```json
{
    "video": {
        "magic": "APF_VER_1",
        "scaler_modes": [
            {
                "width": 240,
                "height": 160,
                "aspect_w": 240,
                "aspect_h": 160,
                "rotation": 0,
                "mirror": 0
            }
        ],
        "display_modes": [
            {
                "id": "0x41"
            },
            {
                "id": "0x42"
            }
        ]
    }
}
```