spiritualized1997/openFPGA-GBA #16: rtc is not working

DR-Soaps, 2022-08-01:
games that need rtc will not work after fist start

PriMieon, 2022-08-01:
Will the game itself not work or the rtc?

DR-Soaps, 2022-08-01:
Just not retaining rtc after reboot

Kitocco, 2023-02-26:
I can add to this; I just started Pokemon Emerald and the setup works fine on launch, but as soon as you leave the game it stops taking time. At the moment it seems the Pocket's clock feature only keeps time for Memeories & screenshot time & dates.

Is there any way for these cores to hook into or read the time passage of the system, or is this something only carts with batteries can do?

evie-gh, 2023-02-26:
Yes, cores can access the RTC, they just need to be updated to do so. I'm
not actually sure if there's going to be any additional development on
these cores, though... radio silence since the first release AFAIK

Guzaboru, 2023-07-02:
@spiritualized1997 
I would love to see the RTC feature for GB/C and GBA core. Analogue implemeted the RTC for the pocket over a year ago.
Are there any plans for additional development or is it the final release of these cores?

k5jat, 2023-10-16:
Sadly, it's mid-October 2023, and still no RTC support. :(

StetsonFinch, 2023-10-20:
@spiritualized1997 I’ll personally pay you 50 dollars to implement RTC. Just hit me up when you do. I’m not joking.

carlosalban, 2023-12-15:
@spiritualized1997 happy to help contribute if you decide to open source.

k5jat, 2024-01-11:
There is a port of the mister gb/gbc cores to the AP. I don't know if the gba core is in the works. You can find the post to r/AnalpoguePocket on Reddit.

awolowiecki720, 2024-12-18:
Any updates?? 

I and many others in the community would love to support RTC for this GBA core