spiritualized1997/openFPGA-GBA #17: GBA ROMs not showing up

Sh1ro, 2022-08-01:
I added around 50 GBA ROMS to the common directory. More than half of them are not showing up on the AP.

SegaSnatcher, 2022-08-02:
> I added around 50 GBA ROMS to the common directory. More than half of them are not showing up on the AP.

Are they unzipped?

Sh1ro, 2022-08-02:
Yes, all are ROMs are unzipped.

bradrevans, 2022-08-02:
I want to add that I have seen this issue on the GB/GBC core also. Sometimes ROMs simply don't show up.

SegaSnatcher, 2022-08-02:
> Yes, all are ROMs are unzipped.

Post a pic of the file layout of your SD card.   Also, try a different SD card and see if it still happens.

bradrevans, 2022-08-02:
> I want to add that I have seen this issue on the GB/GBC core also. Sometimes ROMs simply don't show up.

I have just noticed, in my case, at the bottom of the ROM list there is one additional entry that states: "[Incomplete. Too many/long files]"

Not sure what the limits are here or how it decides which not to show.

SegaSnatcher, 2022-08-02:
> I have just noticed, in my case, at the bottom of the ROM list there is one additional entry that states: "[Incomplete. Too many/long files]"
>
> Not sure what the limits are here or how it decides [...]

Oh right, I totally forgot about that.   If the names of your roms are too long then they won't show up.  Thanks for the reminder. 

I'm not sure what the limit is, but for the games not showing up just shorten the names as much as you can.

bradrevans, 2022-08-02:
> 

I would be inclined to agree with this but I simply removed some ROMs from the folder and others that weren't previously showing up now show. I don't think this is a ROM name length issue as the system supports ellipses. Changing the number of the ROMs in there though did change what was showing up.

bradrevans, 2022-08-02:
> Oh right, I totally forgot about that. If the names of your roms are too long then they won't show up. Thanks for the reminder.
>
> I'm not sure what the limit is, but for the games not showing up just [...]

I would be inclined to agree with this but I simply removed some ROMs from the folder and others that weren't previously showing up now show. I don't think this is a ROM name length issue as the system supports ellipses. Changing the number of the ROMs in there though did change what was showing up.

SegaSnatcher, 2022-08-02:
> EDIT: Can confirm also that upon loading the GBA Core, none of my ROM's prior to the Letter F show and none following the letter T. There is clearly some kind of trimming going on here. A power cycle doesn't change the cores that show and I have also seen examples where, for example, games beginning with the letter P are showing, but not all of them. Pokemon Silver is there but Pokemon Red isn't even though I know both files work as I have backed them up from my working physical carts and tested in an emulator. Deleting Pokemon Silver then triggered Pokemon Red to show up when I next checked.

Weird, not experiencing this at all.   I have my GBA roms in folders.  Show me your rom structure on your SD card.

This is why mine looks like for GBA roms. 

[image]

bradrevans, 2022-08-02:
> > EDIT: Can confirm also that upon loading the GBA Core, none of my ROM's prior to the Letter F show and none following the letter T. There is clearly some kind of trimming going on here. A power cycle doesn't change the cores that show and I have also seen examples where, for example, games beginning with the letter P are showing, but not all of them. Pokemon Silver is there but Pokemon Red isn't even though I know both files work as I have backed them up from my working physical carts and tested in an emulator. Deleting Pokemon Silver then triggered Pokemon Red to show up when I next checked.
> 
> Weird, not experiencing this at all. I have my GBA roms in folders. Show me your rom structure on your SD card.
> 
> This is what my folder structure looks like for GBA roms.
> 
> [image]

I think that explains it then! I had all of my ROMs in one folder, no breakdown such as yours. I've gone ahead and split all of the items out using a similar folder structure to yourself and it everything seems to be appearing now! I guess there is just a limit on how long a list can be and breaking the ROMs out into smaller folders has addressed that. Thanks for your help!

scottmogcrx, 2022-08-05:
I'm not seeing where Sh1ro responded and have another possible reason. I am brand new to this site so hopefully I don't do something stupid.

My GBA roms needed to have a .gba file extension and not a .GBA file extension. The capitalization may be an issue which is exactly the problem that I had at first (6 or 7 days) when I installed the cores.

Guavanon, 2022-08-09:
Just to add to the possible reasons... it seems like it can't recognize any files with a "#" in the name, so those ROMs are just omitted from the list.

Which is annoying since a lot of ROMsets use that character for tagging purposes.