spiritualized1997/openFPGA-GBA #18: Rumble Pak not working with patched games like Pokemon Pinball

stfj, 2022-08-06:
The Rumble Pak works well with Drill Dozer, but it doesn't seem to work with games that have been patched post release to work with rumble packs, like Pokemon Pinball: Ruby and Saphire — https://www.romhacking.net/hacks/6256/

Xukkorz, 2022-08-30:
Confirming this doesn't work. Games that seem to have gameboy player rumble don't work with rumble. Seems to imply this list is hard coded rather than looking at the game rom/header to determine functionality. I presume this implies that homebrew games and rumble added to other games would not work either

missionfloyd, 2022-12-11:
You can make it work by opening it in a hex editor and replacing the title ID at 0xAC with that of a supported game (Drill Dozer is V49E), then fixing the header with [GBATA](https://www.romhacking.net/utilities/601/).

Xukkorz, 2022-12-11:
Yes it functions but this isn't the right way to fix the problem. Community should be able to override this game list for when homebrew wants to use more tricks. What if this method causes the game to not work as intended because different hardware than is expected it loaded. Then we're not able to go this route.

TurtleP, 2023-09-06:
I bought a DS Rumble Pak mainly for this. Tried to play Pokémon Pinball as well and it just doesn't work. I stumbled upon this issue and hope that maybe it'll be fixed one day. At least it's a suitable dust cover for now.

mattdog1000000, 2023-11-27:
> I bought a DS Rumble Pak mainly for this. Tried to play Pokémon Pinball as well and it just doesn't work. I stumbled upon this issue and hope that maybe it'll be fixed one day. At least it's a suitable dust cover for now.

As a temporary workaround I added hackfixes to be applied on top of the existing rumble patches if you're still interested and unable to get rumble working: https://www.romhacking.net/hacks/8252/