spiritualized1997/openFPGA-GBA #31: GBA Core Save Creation Error

LauraKBuzz, 2022-12-10:
Hi.

I installed this core today on my analogue pocket, and while it runs games perfectly, it seems unable to save. For many games I can simply use save states, but games like Minish Cap, where you need to create a save file to start the game, do not work. I get told the save files are corrupted, and an error when I try to enter my player name that it cannot save.

Any ideas?

Nyegnav, 2023-02-03:
Same issue. Any workaround? No game will accept a save.

whitn152, 2023-08-15:
I think this is an issue with the core being able to read the entire rom. I had an issue with my rom of Minish Cap in this core, but it was playing fine on Visual Boy Advance emulator. After deleting some of the data at the end of the rom in a hex editor, it started working on the Pocket.

DunhamGitHub, 2023-08-15:
> Hi.
>
> I installed this core today on my analogue pocket, and while it runs games perfectly, it seems unable to save. For many games I can simply use save states, but games like Minish Cap, where you [...]

I mean ...the bare minimum is to share ROM information, no!?

- _There are around 20 "Minish Cap" ROMs in common circulation, many of them are NOT "verified, good dumps"_

DunhamGitHub, 2023-08-15:
> ? No game will acce

If "no game" can save, it's a hardware issue, not software. Get your Pocket serviced👩🏼‍⚕️