spiritualized1997/openFPGA-GBA #35: Cannot Load Save States and normal Saves become corrupted with specific game

triengage, 2022-12-20:
Using this openFPGA core (version 1.2.0), I ran into issues where a Save State fails to load and also corrupts the normal in-game Save. I can duplicate this every time, however, I've only found the issue to occur on one game so far (Shining Soul II). To duplicate,

1. Get to a point in the game where you make an in-game save, and save
2. Reboot
3. Load up the game in-game
4. Make a Save State
5. Then make another in-game save
6. Reboot again
7. Load the game, but instead of loading from the in-game menu, attempt to load the Save State

After these steps, I receive a red load failure with the Save State, and then I have to reload the game. Then the normal in-game save will have been wiped and all progress lost.

I have attempted these steps with other games like Zelda Minish Cap and Final Fantasy 1+2 Dawn of Souls, but I have not found another game that fails to load the save state and corrupts the in-game save.  

MD5: f4a655e23638e79eacf44a456faee6a4

Mycah, 2022-12-22:
With Metroid Fusion and Metroid Zero Mission I can use sleep/restore about 80% of the time, 20% of the time the core restarts when trying to wake up.
With Mother 1+2 the opposite is true.  I can use sleep/restore about 20% of the time, 80% of the time the core restarts.
I don't know if it matters, but I'm using a Q4 2022 console with a Sandisk Extreme Plus SD Card.

I have asked a friend who has a 2021 console to test Mother 1+2 and he states he doesn't have the same issue with save states sleep/restore.  He has a slower/older Samsung SD card.

tdebug, 2023-10-15:
Memories never work in Sonic Advance, they save but never load. But I believe we should not expect any new version of this core anymore, not at all from this developer - all his cores look long time abandoned.