spiritualized1997/openFPGA-GBA #36: Light sensor use for the Boktai trilogy?

JayBee-git, 2022-12-26:
Hi,
I was wondering if it would be possible to add support to the Botkai series solar sensor.
It's possible to read their solar sensor data through the cartridge port, function that was used in the DS game Lunar Knights, sequel to the series. So maybe there could be a way to use one real boktai cart to have the solar function for the whole trilogy?

Kitocco, 2023-08-25:
I'd like to second this, especially with the existence of the open source cart extender (https://github.com/v1605/open-source-cart-extender), and the Pocket Player Extender, it would be really handy. Especially with English carts becoming so expensive and the 3rd game never getting an official translation, it would be really helpful in allowing people to experience those games as they were meant to be played.

Thank you!

firecrafts, 2023-09-11:
would be cool to have passthrough accelerator  / solar sensor from original cartridge game and pass it to use it with other games using the core , Like :-
- Using translated version of boktai  rom with original cartridge that inserted inside Analogue pocket 
- using Yoshi Topsy Turvy rom with Warioware Twisted  cartridge that inserted inside Analogue pocket  etc