spiritualized1997/openFPGA-GBA #39: Don't execute any game

Bieno64, 2023-06-24:
I have a little problem that started a few days ago and I don't understand why
The order I do is this
openFPGA/All/Gameboy Advance/Run
and then I select the game (with GBA extension) and instead of executing the game, it goes back a directory (common). I go enter again the game directory and now it is empty.
To get the game list to reappear, I have to go back to the RUN command, after selecting the console type.
I've tryed this this a lot of games that worked before.
A few days ago it worked all ok and the only thing I did was saving more ROMs from other systems in their directories. Nothing else
Since it didn't work, I downloaded the files again and saved them over ( https://github.com/spiritualized1997/openFPGA-GBA/releases/download/v1.2.0/Spiritualized_GBA_1.2.0_2022_08_25.zip ) but it doesn't work either.
What am I doing wrong?
Thank you very much for this amazing work !!!!!!!!!

Bieno64, 2023-06-28:
I've done a Global Reset with no luck. It doesn't load any game again
I've tryed to put into the game folder only a few games and again no luck...  :-(
Any hint ?

anamegoeshere23, 2023-08-25:
Did you add the Bios?

Bieno64, 2023-08-26:
Hi

Yes, thanks. Now it  worked well. I use *Matt Pannella's Updater Utility*
and it worked all perfect

Thx anyway

Bieno Martí Braitmaier
[image: https://www.commodoreplus.org/] <https://www.commodoreplus.org/>[image:
https://onslaught.c64.org/] <https://onslaught.c64.org/>[image:
http://www.exploracommodore.com/] <http://www.exploracommodore.com/>[image:
https://www.youtube.com/watch?v=ZzvK5uLu7F0]
<https://vegano.club/por-que-ser-vegano/>