spiritualized1997/openFPGA-GBA #42: Crash on Sword of Mana

anamegoeshere23, 2023-08-24:
Graphical glitches followed by freezing and intermittent audio screeching in the cutscene after starting game and choosing character Girl.

execution101, 2023-09-03:
First I want to say thank you for making this core. I hope in time you can make some tweaks to it.
I would also like to add to this list
-Mother 3 English translation 1.2 and 1.3 crash before even getting through the first cut scene loud audio noises start when it crashes and the screen freezes.

azurehours, 2023-09-15:
Having this problem as well with multiple games. Metroid Zero Mission, Golden Sun: The Lost Age, Rebel Star, etc.
Seems to crash on more games than not. Other cores I've tried have had no issue.
Have a GITD edition.

DunhamGitHub, 2023-09-15:
(Maybe I fail to see after reading all these posts here 3 times) ...do you guys use physical cartridges or the Cores?

* because if it is cores, it's probably related to ill-formatted / cheap SD-cards. Also in this case you could have (should have) dropped the CRC32 of some ROMs do others (like me) can check them...

azurehours, 2023-09-15:
The core for me. I've used a SanDisk Extreme Pro 256GB (UHS-I, A2) for my SD card. Reformatted in ExFAT, did a clean core install, same deal
CRC32s for the mentioned ROMs, for anyone interested

Metroid Zero Mission:
5c61a844

Rebelstar: Tactical Command:
fb6c590f

Golden Sun: The Lost Age:
606a1c4d

DunhamGitHub, 2023-09-15:
> The core for me. I've used a SanDisk Extreme Pro 256GB (UHS-I, A2) for my SD card. Reformatted in ExFAT, did a clean core install, same deal CRC32s for the mentioned ROMs, for anyone interested
> 

Thanks, **ExFAT is notoriously unstable**. (Which is why, as a popular example, the Nintendo Switch SD is usually FAT formatted, not ExFAT).
You only need ExFat if you use ROMs that are over 4GB in size. Since ROMs for this product are usually not over 32mb (unless I have overlooked something).

**FAT format is the way to go for SD on Analogue Pocket.** Please try FAT.

_Question #2:_ Did you try a second SD-Card (newly formatted to FAT)? If not, do so.

azurehours, 2023-09-15:
Reformatted to FAT and still no luck, crashing/freezing in under a minute. I don't have a free second SD card on hand, but I'll order one and report back.

execution101, 2023-09-15:
I wish FAT32 would be the solution. I tried it just now same issue with it crashing in Doom 2. I have a friend that has an Analogue Pocket as well he got his in one of the first shipments that were sent he was the motivator for me to get one after trying his out. Anyways I recorded a video of both of our pockets using my sd card showing that his would work just fine while mine would not work. Here's a link to the video https://www.youtube.com/watch?v=m2cUZZxfq-I

I'm thinking my unit is defective or these newer pockets have some kind of hardware revision.

azurehours, 2023-09-19:
Tried a new microSD, a SanDisk Extreme 32GB, same problem. Something I've noticed is that some games will freeze and the audio continues as normal, while others will freeze and the audio starts glitching.

anamegoeshere23, 2023-09-21:
I don't think it's an sd card issue, I tried a couple cards and a number of configurations, all the same result.

execution101, 2023-09-21:
Yeah I don't think its an SD card issue as well. Either the core needs some updating do to a hardware revision or some of the pockets being shipped out are defective in some way with the hardware. I've been reaching out to Analogue Customer Service and they are so slow at responding.

MarcoPNS, 2023-10-09:
Still in contact with the support. Unfortunately, no result so far. Neither Core nor Catridge GBA is playable for me. The transparent version will be coming soon and I will test it directly.

execution101, 2023-10-11:
https://www.reddit.com/r/AnaloguePocket/comments/16jhd6n/comment/k3vrekj/?utm_source=share&utm_medium=mweb3x&utm_name=mweb3xcss&utm_term=1&utm_content=share_button

So this guy had a suggestion of using the 1.0 release of the core. I just put the 1.0 release on and so far with Doom 2 and Castlevania I haven't experienced any crashes. I didn't add this to my post since I hadn't tried playing it yet but playing Kirby and the Amazing Mirror it would crash after a few minutes of play. On the 1.0 I haven't had crashes in that Kirby game as well.

Can you guys try and see if that works for you?

MarcoPNS, 2023-10-11:
I can confirm that v1.0.0 does not crash. I don't know why. I tried to compare the two versions and maybe because v1.2.0 does not have "variants"? Or maybe some of the options are buggy? Idk.

jakubito, 2023-12-10:
I have the same issue with version 1.2.0, yellow Pocket. Final Fantasy 6 crashes right after starting a new game. Can confirm downgrading to 1.0.0 helped

Computerlabrat, 2025-08-22:
It also has an issue with the Sonic SatAM fan game Sonic Epoch Advance. After you delete the existing save files and play, it seems it won't move on to level 2.