spiritualized1997/openFPGA-GBA #44: Crashes on GITD pocket, but not launch pocket.

Charlie22911, 2023-09-15:
I’m seeing a new issue that only exists on the glow in the dark version of the pocket, when games will crash. Great example is F-Zero: Maximum Velocity, it crashes at the start of a race. It works fine on the launch pocket.
I’ve checked the rom CRC, and it matches the dump of my physical game. Also tried different SD cards.

dallowbg, 2023-09-15:
Strange issue. Tested that that game on my GitD pocket and it works fine. No crash at the start of the race.

azurehours, 2023-09-15:
Same problem here across various games, hoping it's a core or firmware problem and not something wrong with the second FPGA

Charlie22911, 2023-09-15:
Here is a GBA test suite on both pocket editions, showing clear differences in results. Don’t have a lot of free time this weekend, but I’ll try to document more.

https://ibb.co/zVnSD0w
https://ibb.co/68wD7qk

dallowbg, 2023-09-15:
> Here is a GBA test suite on both pocket editions, showing clear differences in results. Don’t have a lot of free time this weekend, but I’ll try to document more.
> 
> https://ibb.co/zVnSD0w https://ibb.co/68wD7qk

I believe I just ran the same test but they all show as Pass for me.

https://i.imgur.com/0L5dOlI.jpg

Charlie22911, 2023-09-15:
> I believe I just ran the same test but they all show as Pass for me.
>
> https://i.imgur.com/0L5dOlI.jpg

Interesting, I’ve noticed that the “Start” test will sometimes pass on my GITD with the Spiritualized core. “Flip 1” always fails, on both pockets and my AGB-001 with AGS screen.

Charlie22911, 2023-09-15:
Photo of crash, because it’s dawned on me I’ve omitted it:
https://ibb.co/pjmRDXC

MarcoPNS, 2023-09-16:
I have the same Issue with the white Pocket. Would assume that it is a similar issue to #42 - maybe a hardware issue with some batches not limited to a specific color. my GITD works fine for example.

anamegoeshere23, 2023-09-26:
It's interesting that it appears to only seems to be spiritualized gba that is affected

edrethardo, 2023-10-18:
I have the same issue with my transparent orange pocket

sagisarius, 2023-11-01:
Same issue.  Transparent purple model.  Doesn't really matter which ROM I play, it crashes.  Reverting to 1.0 stops the crashing.  There are several reddit threads of folks trying to figure this out as well.

burim27, 2023-12-26:
As posted on reddit, I have the same issue with a gba game.  I’ve noticed that when my analogue pocket reaches the temperature of 39°C, it freezes and I need to let it cool down till I can play again. Hope it helps.

MarcoPNS, 2023-12-26:
@burim27 the temperature can't be the issue. 39C is pretty low. Already tried to downgrade the core to 1.0?

burim27, 2023-12-27:
@MarcoPNS can't put my hands on core 1.0. Do you have any idea where to find it ? I only find the 1.2 one

burim27, 2023-12-27:
@MarcoPNS thanks for the advice, it works with the 1.0 one and the temperature can go above 39C. I've done multiple tests with core 1.2 and the game was crashing each time the analogue reached 39C

edrethardo, 2023-12-27:
> @MarcoPNS thanks for the advice, it works with the 1.0 one and the temperature can go above 39C. I've done multiple tests with core 1.2 and the game was crashing each time the analogue reached 39C 

Does your Pocket reach 39C after less than 5 minutes?
Mine crashes when playing Pokémon pinball ruby & Saphire in a very short amount of time.

burim27, 2023-12-27:
@edrethardo I’m not sure how much time it was taking till it reached 39° but I know that with Kirby and the amazing mirror it was crashing very fast. With core 1.0 I noticed that it doesn’t get hot that fast and doesn’t crash so maybe it could help

EvilPixelsKill, 2023-12-28:
My pocket doesn't like 1.0 either. Especially playing Iridion II.