spiritualized1997/openFPGA-GBA #45: Frequent crashes on Analogue Pocket with games that works for other users

mikeyvids, 2023-10-21:
Unsure if there are variations in the Pocket hardware but I'm get frequent crashes at the beginning of games using this core (see videos)
https://youtube.com/shorts/sIPGKLMpkVY?feature=share
https://youtu.be/6i9TB7GzGd0
https://youtube.com/shorts/Sg_h4opJapY?feature=share

I tried 3 microSD cards multiple times reformatted to Exfat and FAT32. I used pocket_updater and Pocket Sync. I've tried multiple games, and multiple sources of the same games, all experiencing crashes around the same area of each game.

Outside of some audio playback issues in PCECD and black screen for Genesis romhacks all other cores seem to be working fine on openFPGA. I hear many people do not have this issue at all, so I'm hopeful that its a hardware variation it can be addressed in upcoming updates.

k5jat, 2023-11-17:
Hi! I'm not the dev, but thought I might try and help. Is your crashing issue still happening?

razor85, 2023-12-17:
> Hi! I'm not the dev, but thought I might try and help. Is your crashing issue still happening?

mine is

k5jat, 2024-01-01:
For those who are having this issue, try using roms from the Tiny Best Set. I have used them on every handheld I've had and they seem to work well.

razor85, 2024-01-01:
Still same problems :/

divanikus, 2024-10-16:
Having this issue with Metroid Zero Mission as well. Works fine from cart