spiritualized1997/openFPGA-GBA #49: Whitelisting Goodboy Galaxy for rumble

exelotl, 2023-12-02:
Hello!

[Goodboy Galaxy](https://goodboygalaxy.com/) has a rumble feature which works in the same way as Drill Dozer (both GB Player rumble and GPIO3 rumble are supported)

Sadly, the rumble pak support of this openFPGA core doesn't work for Goodboy unless the user patches the Game Code / ROM ID to match that of a different rumble game (such as Drill Dozer).

Would it be possible to add Goodboy Galaxy's ID (`2GBP`) to your list so that rumble works out of the box?

------

p.s. we also had several reports of crashes on the Pocket due to issue https://github.com/spiritualized1997/openFPGA-GBA/issues/44 - would be very nice if that could get fixed too!

mattdog1000000, 2023-12-09:
There is a workaround patch for getting rumble to work on Analogue Pocket provided by the devs here: https://goodboygalaxy.com/wiki/doku.php?id=rumble

Aero2946, 2024-01-04:
Please also whitelist Apotris! It is another homebrew GBA game that supports rumble.

I'm sorry, I don't know the ROM ID, but the .gba ROM is available here: https://akouzoukos.com/apotris