spiritualized1997/openFPGA-GBA #54: Crashing on 2023 Pockets

EvilPixelsKill, 2023-12-28:
Both releases of this core [1.0.0] [1.2.0] are crashing on my August 23 delivered Analogue Pocket. We are all aware of this happening on the new Pockets and I just wanted to add my gratitude for the core in the first place and request some investigation as to what might be going on? 

Is it the core[s] or hardware, Thank You!

ctafield, 2024-01-08:
Still very much happening, even after the 2.0 firmware. Pretty much unusable as a core, unfortunately.

EvilPixelsKill, 2024-01-08:
Awesome, thanks for the info. I was getting worried it was the pockets hardware. Hopefully we get a fix one day.

notmayo, 2024-01-18:
Neither Golden Sun game will work for me, even on yesterday's (2.1) newest firmware. The game sometimes starts but makes audio glitch sounds and then freezes. The pocket itself is not frozen. I've switched SD cards, format type, bios files, verified roms, etc.

ctafield, 2024-01-18:
Also just tried 2.1 on my glow in the dark pocket.

For the most part, pretty much everything still crashes either immediately, or after a very short while. 
Duke Nukem Advance used to get in game, but now it just crashes on the first screen with the intro text/copyrights.

EvilPixelsKill, 2024-01-28:
Can anyone confirm if the games work flawlessly using a everdrive type cart? We could all rule out a AP hardware issue then.

javascriptjoey, 2024-02-04:
I just received My Analog Pocket and also are experiencing consistent crashes trying to load any GameBoy Advanced games  Ive tried both fw 2.0 and 2.1 and neither made any difference in this result. Also tried replacing the spiritualized core files with what is distributed directly from this repository and that had no positive effect on the issues currently faces trying to play  GBA games the result is the same the games crash on boot or lock and freeze with an audio glitch sound

javascriptjoey, 2024-02-22:
Any updates on this open issue?

notmayo, 2024-04-16:
The only way I can get games like Golden sun and FF6 working is to delete the Spiritualized.GBA folder from SDCARD\Roms folder and copy over the 1.0 release to the SD card. Now these games are letting me get past the freezing areas that I kept experiencing.