spiritualized1997/openFPGA-GBA #62: Core does not support mGBA RTC .sav files

keegandahm, 2024-04-21:
mGBA appends the 16 byte realtime clock value to its .sav files ([source code here](https://github.com/mgba-emu/mgba/blob/61791c91cb41eaafd4a710f8c6229840e5a343b4/src/gba/savedata.c#L591)). Loading those .sav files in the Spiritualized GBA core fails as the file is too large, since it doesn't exactly match the expected SRAM chip size.

This was previously mentioned in issue #37 but closed shortly after by the reporter who opened it. I wanted to make this issue just to be certain the dev sees it, and hoping that having more info available makes this something you can support, or explicitly avoid supporting.

And last, a question: how does this core handle RTC currently? Does it use Pocket's internal clock?

Thanks for all the work you did making this core, it's a lot of fun!

## For people hitting this from Google

If you get the error "`Error in framework: File ID [nnn] too large`", you probably copied a .sav file from an emulator on your PC to your Analogue Pocket's SD card, and you used an emulator (like mGBA) that adds the realtime clock's value to the .sav files it writes.

According to the first issue, you can use the mGBA tool under `File` > `Save games` > `Convert save game...` to do this, but for some reason that tool didn't work for me.

If you use macOS or Linux, you can accomplish the same thing by first backing up your .sav file, then running `truncate -s -16 /path/to/your/game.sav`. This removes the last 16 bytes from the end of the file. I'm not aware of an easy alternative on Windows.

jjkraus01, 2024-07-06:
This was my issue, thanks for the pointing this out! Had the exact problem.
For me, on Windows I just upgraded to the latest mGBA version and was able to convert my file.

PilotJeremiah, 2025-02-12:
On Mac, updating mGBA and using the covert file also worked for me. Strangely enough the .sav came from an authentic cart dumped via gb operator but my pocket wasn't having it. Works fine after converting.